intel/FSP #5: Broadwell-DE FSP

zaolin, 2018-11-06:
@nate-desimone-intel  Currently headers and sample code are missing in the repository. Can you add the missing ones?

zaolin, 2018-11-06:
Also maybe update them?

nate-desimone-intel, 2018-11-06:
Hi @zaolin, I see the following: https://github.com/IntelFsp/FSP/blob/master/BroadwellDEFspBinPkg/include/fspvpd.h

Is there something more you are looking for?

zaolin, 2018-11-06:
Yep @nate-desimone-intel https://review.coreboot.org/cgit/coreboot.git/tree/src/vendorcode/intel/fsp1_0/broadwell_de/include
https://review.coreboot.org/cgit/coreboot.git/tree/src/vendorcode/intel/fsp1_0/broadwell_de/srx

nate-desimone-intel, 2018-11-06:
Hi @zaolin, when comparing the contents of both repos, they are identical:
https://github.com/IntelFsp/FSP/tree/master/BroadwellDEFspBinPkg/include
https://review.coreboot.org/cgit/coreboot.git/tree/src/vendorcode/intel/fsp1_0/broadwell_de/include

[image]

The copy of fspvpd.h in github is newer than the version in the coreboot repo:

[image]

The fspsupport.c file is also present in github, just under a different directory structure. I am closing this issue.

closed 2018-11-06 by nate-desimone-intel