intel/FSP #6: FSP 1.0 headers throwing warnings

zaolin, 2018-11-06:
@nate-desimone-intel Intel FSP 1.0 headers throwing warnings which are interpreted as errors in coreboot. It would be great if it gets fixed.

Output:
`src/drivers/intel/fsp1_0/hob.c: In function 'print_hob_type_structure':
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:318:6: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ((*(EFI_HOB_GENERIC_HEADER **)&(HobStart))->HobType)
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:361:37: note: in expansion of macro 'GET_HOB_TYPE'
 #define END_OF_HOB_LIST(HobStart)  (GET_HOB_TYPE (HobStart) == (UINT16)EFI_HOB_TYPE_END_OF_HOB_LIST)
                                     ^~~~~~~~~~~~
src/drivers/intel/fsp1_0/hob.c:202:13: note: in expansion of macro 'END_OF_HOB_LIST'
   Lasthob = END_OF_HOB_LIST(Currenthob); /* Check for end of HOB list */
             ^~~~~~~~~~~~~~~
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:346:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   (VOID *)(*(UINT8 **)&(HobStart) + GET_HOB_LENGTH (HobStart))
             ^~~~~~~~~~~~~~~~~~~~~
src/drivers/intel/fsp1_0/hob.c:204:14: note: in expansion of macro 'GET_NEXT_HOB'
    Nexthob = GET_NEXT_HOB(Currenthob); /* Get next HOB pointer */
              ^~~~~~~~~~~~
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:332:6: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ((*(EFI_HOB_GENERIC_HEADER **)&(HobStart))->HobLength)
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:346:37: note: in expansion of macro 'GET_HOB_LENGTH'
   (VOID *)(*(UINT8 **)&(HobStart) + GET_HOB_LENGTH (HobStart))
                                     ^~~~~~~~~~~~~~
src/drivers/intel/fsp1_0/hob.c:204:14: note: in expansion of macro 'GET_NEXT_HOB'
    Nexthob = GET_NEXT_HOB(Currenthob); /* Get next HOB pointer */
              ^~~~~~~~~~~~

src/drivers/intel/fsp1_0/fsp_util.c: In function 'save_mrc_data':                       
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:375:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   (VOID *)(*(UINT8 **)&(HobStart) + sizeof (EFI_HOB_GUID_TYPE))                     
             ^~~~~~~~~~~~~~~~~~~~~                                                 
src/drivers/intel/fsp1_0/fsp_util.c:255:17: note: in expansion of macro 'GET_GUID_HOB_DATA'
  mrc_hob_data = GET_GUID_HOB_DATA (mrc_hob);                                                    
                 ^~~~~~~~~~~~~~~~~                                                    
3rdparty/fsp/RangeleyFspBinPkg/Include/fsphob.h:332:6: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   ((*(EFI_HOB_GENERIC_HEADER **)&(HobStart))->HobLength)                                  
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                 
src/drivers/intel/fsp1_0/fsp_util.c:256:23: note: in expansion of macro 'GET_HOB_LENGTH'                                                                                                                                                     
  mrc_hob_size = (u32) GET_HOB_LENGTH(mrc_hob);                                                                                                                                                                                              
                       ^~~~~~~~~~~~~~                
`

hannahwilliams2, 2019-04-16:
@nate-desimone-intel 
I am also seeing this same build error. Replacing with fspvpd.h from previous version makes this build error go away. Can you please fix this.

veusebio, 2019-04-18:
@hannahwilliams2 which version worked?

hannahwilliams2, 2019-04-22:
@veusebio 
Sorry I actually meant replacing fsphob.h from older version worked (not fspvpd.h)
This was the diff I was referring to - 
318c318
<   ((*(EFI_HOB_GENERIC_HEADER **)&(HobStart))->HobType)
---
> 	(((EFI_HOB_GENERIC_HEADER *)(HobStart))->HobType)
332c332
<   ((*(EFI_HOB_GENERIC_HEADER **)&(HobStart))->HobLength)
---
> 	(((EFI_HOB_GENERIC_HEADER *)(HobStart))->HobLength)
346c346
<   (VOID *)(*(UINT8 **)&(HobStart) + GET_HOB_LENGTH (HobStart))
---
> 	(VOID *)((UINT8 *)(HobStart) + GET_HOB_LENGTH(HobStart))
361c361
< #define END_OF_HOB_LIST(HobStart)  (GET_HOB_TYPE (HobStart) == (UINT16)EFI_HOB_TYPE_END_OF_HOB_LIST)
---
> #define END_OF_HOB_LIST(HobStart)  (GET_HOB_TYPE(HobStart) == (UINT16)EFI_HOB_TYPE_END_OF_HOB_LIST)
375c375
<   (VOID *)(*(UINT8 **)&(HobStart) + sizeof (EFI_HOB_GUID_TYPE))
---
> 	(VOID *)((UINT8 *)(HobStart) + sizeof(EFI_HOB_GUID_TYPE))
388c388
<   (UINT16)(GET_HOB_LENGTH (HobStart) - sizeof (EFI_HOB_GUID_TYPE))
---
> 	(UINT16)(GET_HOB_LENGTH(HobStart) - sizeof(EFI_HOB_GUID_TYPE))