intel/FSP #7: FSP tooling only support FSP 2.0

zaolin, 2018-11-06:
The FSP tooling which is needed for rebasing only supports FSP 1.0.
Is there a way that Intel adds 1.0 and 1.1 support as well?

zaolin, 2018-11-06:
@nate-desimone-intel Because currently a lot FSP 1.0 and 1.1 related stuff are available with coreboot.

nate-desimone-intel, 2018-11-06:
Hi @zaolin, you are correct that SplitFspBin.py does not work for FSP 1.x binaries. For these older FSPs, one must use BCT to do the rebase:

https://github.com/IntelFsp/BCT

I realize that this is not ideal. I recommend that we put a feature request to have SplitFspBin.py support FSP 1.x binaries in the tianocore bugzilla: https://bugzilla.tianocore.org/ Would you like me to file this request on your behalf, or would you prefer to do so?

zaolin, 2018-11-07:
If you have an account, please do it.

nate-desimone, 2020-01-07:
Fixed in 0bc2b07e.

closed 2020-01-07 by nate-desimone