intel/FSP #10: BroadwellDEFspBinPkg MR2 locks IA32_FEATURE_CONTROL in FSP-M

PatrickRudolph, 2019-01-08:
The IA32_FEATURE_CONTROL MSR is locked on all cores after memory init, but it should be locked in FspNotify(EnumInitPhaseReadyToBoot).
That would allow coreboot to set sane values, instead of the FSP default of 0x4.

rminnich, 2019-01-09:
that should never be locked by anything but the kernel. Anything else is a mistake. We went through this seven years ago on chromebooks, I can't believe we're back here.

PatrickRudolph, 2019-02-04:
@nate-desimone Any update on this issue ?

shwong1, 2019-02-14:
The issue will be investigated, you may contact your Intel rep or file an IPS (Intel Premier Support) ticket if it’s urgent for you. Thanks!

PatrickRudolph, 2019-04-05:
It's been a month without a reply. Any update on this issue ?

lsiudut, 2019-05-16:
At this point it's been a months. I'm +1-ning for update on that.

PatrickRudolph, 2019-11-16:
Almost a year! You must be typing very slow as it's only one line of source code to change...

swong23, 2019-12-04:
Root cause : after memory initialization  the FSP programs "IA32_FEATURE_CONTROL" MSR with Vmx features and lock the register by setting 0th bit; once the lock bit set the register become read only.

Proposed solution :  do not lock the register after memory init  instead  lock  the register during FSP ready to boot stage . 

This will be resolved in the next official release however the timeline of next release is still tbc.
For NDA customers and technology partners of Intel, you may contact your Intel Rep or file an IPS# ticket for source request or method consultation for temporary workaround on your own.

lynxis, 2021-11-30:
@swong23 any update on this (next month this will be 3 years old)