intel/FSP #12: Braswell: not working Memory Init

miczyg1, 2019-01-28:
I am experiencing problems with FSP Memory Init on Braswell SoCs. It seems like memory initialization fails because Memory Init is not returning, platform hangs after calling FSP Memory Init.
I have following processors that fail: Celeron J3060 and Celeron J3160; both have the same issue.

The platforms have 1 SODIMM module on channel 0. I am feeding the UPD header with correct SPD, but still no luck. I have also tried FSP MR1 and MR2, but nothing works. 

The memory DIMMs I have:

https://www.samsung.com/semiconductor/dram/module/M471B5173DB0-YK0/
https://www.samsung.com/semiconductor/dram/module/M471B5273DH0-YK0/

Are these modules not compatible with FSP?

pietrushnic, 2019-02-12:
@nate-desimone  @mauricema @swang57 @veusebio @GiriM123 are there any chance anyone from Intel can take a look at this problem? We recfently tried also other modules M471B5273CH0-CK0 and those also have problems. We heard that some developers have access to FSP version with debug logs, is it possible for Braswell?

shwong1, 2019-02-14:
Yes, there's debug version of FSP bin available from Intel. Kindly contact your Intel rep or file an IPS (Intel Premier Support) ticket to request the FSP debug bin.

pietrushnic, 2019-02-14:
@shwong1 thank for that reply, I have to contact my FAE to get access to IPS account. I will get back with feedback if that path worked.

miczyg1, 2019-03-08:
We have managed to get FSP MR2 work on mentioned processors. However, we had to patch the FSP binary with BCT and turn SecureBoot option to `Disabled` instead of leaving default `Auto`. In such condition, the binary cannot be used out of the box from this repository.  How is the `Auto` option working? Is there a way to workaround this problem?

nate-desimone, 2019-04-19:
Hi @pietrushnic,

There are two versions of Braswell FSP, a regular version and a secure boot version. There is a build flag that controls which version of the FSP binary is generated at compile time.

I believe the confusion comes from the fact that the MR1 release gave a regular build whereas the MR2 release gave a Secure Boot build. We have posted updated versions of (1.1.7.0 and 1.1.8.0) and we were careful to give both the regular and Secure Boot versions of the FSP binary this time.

https://github.com/IntelFsp/FSP/tree/master/BraswellFspBinPkg/FspBin
https://github.com/IntelFsp/FSP/tree/master/BraswellFspBinPkg/FspBin/SecureBootEnabled

Given that you have to change the SecureBoot option to `Disabled` it sounds like your bootloader does not support the SecureBoot option. Therefore, it seems like using the regular version of Braswell FSP would be better suited to your needs. Hope that helps.

closed 2019-04-19 by nate-desimone