intel/FSP #14: Braswell: Conflicting FSP versions

i-c-o-n, 2019-03-01:
Hi,

we have again spotted an oddity in coreboot that makes it look like there were forks of FSP. In our repository [1] we have headers for $BSWFSP$ rev 01010700 while here on Github we had $BSWFSP$ rev 01010200 first (MR1?), and now BSWSBFSP rev 01010401 (MR2?).

Is there a chance to get a current $BSWFSP$ into your repository? and in case that it wouldn't cover all features of BSWSBFSP, have both coexist on the same branch?

Best regards,
Nico

[1] https://review.coreboot.org/cgit/coreboot.git/tree/src/vendorcode/intel/fsp/fsp1_1/braswell/FspUpdVpd.h

shwong1, 2019-03-01:
Yes, 01010200 was MR1; then 01010401 was MR2

MrChromebox, 2019-03-01:
@shwong1 right, but the issue is that most (all?) Braswell-based Google ChromeOS devices are compiled against a post-MR2 version (ie, 1.1.7.0), which adds UPDs not present in MR2. Since there's at least one difference in offset, compiling for a non-Google Braswell device is problematic since the header in the coreboot repo doesn't match that for the only publicly available blobs (MR1/MR2).

Ideally there would be a MR3 released which would include the UPDs in the 1.1.7.0 header and allow for all braswell devices to be built using a publicly available FSP header/blob

i-c-o-n, 2019-03-15:
@shwong1 @nate-desimone 
Can we have an update on this? Even a simple "somebody is/is not/will be/won't be looking into this" would already help.

nate-desimone, 2019-04-19:
Hi @i-c-o-n,

$BSWFSP$ is the signature for the regular version of BSW FSP. BSWSBFSP is the signature for the Secure Boot version of FSP.

There is a build flag that controls which version of the FSP binary is generated at compile time.

Hi @MrChromebox,

We have posted updated versions of BSW FSP, both 1.1.7.0 (3f2569c) and the latest and greatest 1.1.8.0 (1d2b7e1).

I am well aware that neither of these versions have an identical UPD header file to the one that is in coreboot right now. I tried very hard to find where that version in coreboot came from. I looked in every source control system on the Intel corporate network that I know about and it is literally nowhere to be found. Intel does share the FSP source code with select partner customers under NDA. It is possible that a 3rd party released an unofficial fork of BSW FSP, but I honestly don't know. The version of 1.1.7.0 posted here is the official release from Intel.

closed 2019-04-19 by nate-desimone