intel/FSP #15: Braswell MR2: Internal UART can't be disabled

PatrickRudolph, 2019-03-06:
Internal UART can't be disabled using UPDs PcdEnableHsuart0/PcdEnableHsuart1.
See [https://review.coreboot.org/#/c/coreboot/+/28464/](https://review.coreboot.org/#/c/coreboot/+/28464/) for reference.

PatrickRudolph, 2019-03-06:
Adding @swang57 @nate-desimone-intel

PatrickRudolph, 2019-04-05:
It's been a month without a reply. Any update on this issue ?

nate-desimone, 2019-04-19:
Hi @PatrickRudolph, we just posted the latest and greatest version, can you give this one a try and see if it helps?

siro20, 2019-04-20:
@nate-desimone Where is it? It's not on the master branch.

nate-desimone, 2019-04-23:
Hi @siro20, yes it is:

https://github.com/IntelFsp/FSP/tree/master/BraswellFspBinPkg/FspBin

Version 1.1.8.0 was posted last Friday, we matched the 1.1.8.0 commit date with the date that the binary was originally compiled.

PatrickRudolph, 2019-05-03:
Bug is still valid as reported in https://review.coreboot.org/c/coreboot/+/28464

PatrickRudolph, 2019-07-10:
It's been a few month. Any update on this issue?

nate-desimone, 2019-07-10:
Hi All,

It looks like this issue was found during development of the firmware for the "Facebook FBG-1701" system. Would it be possible for someone from Facebook whom has a support contract with Intel to file an [IPS](https://www.intel.com/content/www/us/en/design/support/ips/training/access-and-login.html) ticket?

I think the added justification of a customer with a support contract asking for this fix will increase visibility on this issue. Otherwise Intel management is unlikely to allocate engineering time for new firmware development on a 5 year old SoC, especially since you have a workaround.

dhendrix, 2019-07-25:
It seems we'll get a Technical Advisory, but a new FSP will **not** be released to fix this. Perhaps the fix will come in a future FSP release, but for now https://review.coreboot.org/c/coreboot/+/28464 will need to be used as a workaround.

swong23, 2019-11-26:
This shouldn't be an FSP issue:
 The reason why ACPI mode is always enabled is not due to FSP firmware bug. It is a combination of coreboot  ACPI override code and serial console delay.

The FSP is correctly populating PcdEnableHsuart0 and PcdEnableHsuart1. 

The serial console delays its output on first boot that it only displays debug logs during DXE. By that time coreboot has already overridden HSUART1 and HSUART2 in ACPI mode. That is why the logs always report ACPI mode is enabled.

--> To workaround it for coreboot, kindly apply the patch as illustrated as follow:
[image]

 This patch will target only to coreboot  and this can be the permanent fix. 

It has been tested with BCT to check whether or not if the changes will take effect. 
The patch works every time when changing PcdEnableHsuart0 and PcdEnableHsuart1, it is reflected in the logs. In Yocto, it's  verified that HSUART1 is active when enabled.

closed 2019-11-26 by swong23

dhendrix, 2019-11-28:
Thanks @swong23!  I added a comment at https://review.coreboot.org/c/coreboot/+/28464/ to let others know that you have found the root cause.