intel/FSP #19: CherryTrail : Missing support

oopen, 2019-03-29:
CherryTrail are Airmont based, where Airmont is the 14 nm die shrink of Silvermont

Other Airmont FSP support exist : Braswell

When Silvermont have BayTrail FSP

Braswell OR BayTrail can be used for CherryTrail chips ?

shwong1, 2019-03-29:
Neither Braswell nor Bay Trail can be used for Cherry Trail.

oopen, 2019-03-29:
There is no release planned ?

nate-desimone, 2019-04-05:
Not at this time.

closed 2019-04-05 by nate-desimone

nicklasfrahm, 2021-06-10:
I was wondering if it would be possible to reopen this? I bought a Rock Pi X, which has an Intel Atom x5-Z8350 and comes with an AMI UEFI BIOS. The issue however is that this does not support netbooting due to the missing driver for the RTL8168. So I am considering to build EDK II for it, but that would require this FSP.