intel/FSP #25: SpiFlashCfgLockDown UPD vanished after KBL

n-huber, 2019-06-18:
In the later Kaby Lake FSP releases, there used to be a SpiFlashCfgLockDown UPD. Is there any option with similar effect for newer platforms?

Without this option, it seems impossible to let FSP finish its silicon initialization, and use peripherals like the AHCI or xHCI controllers before FLOCKDN (or PRR34_LOCKDN) is set. A bootloader using FSP is then forced to either perform firmware updates in early stages which significantly increases attack surface there, or to implement different boot modes with different locked SPI settings. In any case it seems to increase complexity of the security concept and thus makes it more error-prone.

c0d3z3r0, 2019-10-20:
@nate-desimone ping!

nate-desimone, 2020-03-06:
@n-huber, this UPD has been removed in Coffee Lake and newer platforms. FLOCKDN is done later, specifically during the FspNotifyPhase(EnumInitPhaseAfterPciEnumeration). The boot-loader should make sure that any non-SMM writes to SPI are done before invoking this NotifyPhase().

nate-desimone, 2020-08-28:
I guess I'll assume no news is good news and my comment above answered this question. Closing, if something more is needed please open a new issue.

closed 2020-08-28 by nate-desimone