intel/FSP #26: BroadwellDEFspPkgBin USB initialization issues

MrChromebox, 2019-06-18:
With Broadwell-DE MR2 FSP, USB port initialization has multiple issues:

- only user-facing controls are to enable/disable the two EHCI controllers; nothing for XHCI
- ECHI #2 PCI device is not detected by coreboot regardless of enabled/disabled state
- with SeaBIOS payload, attached USB devices only detected if FSP debug level set to 3 *and* MRC cache disabled (or first boot after flashing)

Ashmita21, 2019-07-22:
What tool are you using for enabling/disabling interfaces like EHCI controllers?

MrChromebox, 2019-07-22:
using the coreboot Kconfig options, which set the UPDs passed to FSP silicon init. En/disable EHCI1, ECHI2 are only available options/UPDs, and function as expected (ie, PCI device disabled/hidden)