intel/FSP #28: FSP: Apollolake: dies in fsp_silicon_init

austriancoder, 2019-07-16:
I am having a wired issue on a custom APL based system. After about  150 warm-reboots the device does not boot up and I pined it to the silicon init call in done by coreboot.
I *think* its related to PCIe as we are switching FPGA vendors and the old one there are no issues. So it might be great to gain access to a debug version of the APL FSP. If it helps the company I work for have NDAs signed with Intel.

swong23, 2019-07-16:
Hi, If you are working for the company that's having NDA's signed with Intel, you may file an IPS# ticket to request for the debug version of the APL FSP bin. Thanks!

shwong1, 2019-07-25:
Closing the issue as no further comment.

closed 2019-07-25 by shwong1