intel/FSP #30: KabylakeFspBinPackage: Crash in FSP

PatrickRudolph, 2019-08-07:
If PchHdaVcType is zero in FSP_SIL_UPD the FSP-S crashes due to NULL pointer dereference.
As the Hda/Azalia device is disabled on my board, I don't see why this needs to be set at all.

c0d3z3r0, 2019-10-01:
Additional info: @PatrickRudolph has a Supermicro X11SSH board. On a very similiar board (Supermicro X11SSM) this does not happen with PchHdaVcType=0

c0d3z3r0, 2019-10-20:
@Intel ping!

nate-desimone, 2019-11-16:
@PatrickRudolph, @c0d3z3r0, please file this in the [Intel® Premier Support](https://www.intel.com/content/www/us/en/design/support/ips/training/access-and-login.html) system and our Application Engineering department will work with you to resolve the issue.

closed 2019-11-16 by nate-desimone

c0d3z3r0, 2019-11-16:
@nate-desimone I don‘t have Premier Support. Please forward this internally

PatrickRudolph, 2019-11-16:
@nate-desimone Please be aware that contributors might not be able to use "Intel Premier Support" as they aren't part of any company or doesn't have an "Intel representative" they could contact.
It makes me sad to see how Intel threads individuals of the Open Source community, by
a. Not responding within month,
b. Closing bug-reports without even confirming that the issue exists,
c. Pointing to unusable internal bugtrackers.
For me your business model looks very questionable. One might think Intel doesn't care if it ships broken (and possibly insecure) software.