intel/FSP #35: BroadwellDEFspPkgBin hiding IIO port for GbE after MemoryInit breaks NotifyStage

anpetrovfb, 2019-10-15:
Looks like hiding IIO port for GbE (0:2.2 and 0:2.3) after MemoryInit causes FspNotify to hang. Current workaround is to do the hiding before MemoryInit

anpetrovfb, 2019-10-15:
this seems to happen only on some SKUs

nate-desimone, 2019-11-16:
@anpetrovfb, please file this in the [Intel® Premier Support](https://www.intel.com/content/www/us/en/design/support/ips/training/access-and-login.html) system and our Application Engineering department will work with you to resolve the issue.

closed 2019-11-16 by nate-desimone

PatrickRudolph, 2019-11-16:
@nate-desimone Please be aware that contributors might not be able to use "Intel Premier Support" as they aren't part of any company or doesn't have an "Intel representative" they could contact.
It makes me sad to see how Intel threads individuals of the Open Source community, by
a. Not responding within month,
b. Closing bug-reports without even confirming that the issue exists,
c. Pointing to unusable internal bugtrackers.
For me your business model looks very questionable. One might think Intel doesn't care if it ships broken (and possibly insecure) software.