intel/FSP #37: Kabylake MRC caching does not work with ECC memory

c0d3z3r0, 2019-10-20:
Kabylake MRC caching does not work with ECC memory

c0d3z3r0, 2019-10-20:
@nate-desimone

c0d3z3r0, 2019-11-15:
@nate-desimone this is an absolute blocker; why is nobody caring??

nate-desimone, 2019-11-15:
@c0d3z3r0, please file this in the [Intel® Premier Support](https://www.intel.com/content/www/us/en/design/support/ips/training/access-and-login.html) system and our Application Engineering department will work with you to resolve the issue.

closed 2019-11-16 by nate-desimone

c0d3z3r0, 2019-11-16:
@nate-desimone I don‘t have Premier Support. Please forward this internally

PatrickRudolph, 2019-11-16:
@nate-desimone Please be aware that contributors might not be able to use "Intel Premier Support" as they aren't part of any company or doesn't have an "Intel representative" they could contact.
It makes me sad to see how Intel threads individuals of the Open Source community, by
a. Not responding within month,
b. Closing bug-reports without even confirming that the issue exists,
c. Pointing to unusable internal bugtrackers.
For me your business model looks very questionable. One might think Intel doesn't care if it ships broken (and possibly insecure) software.