intel/FSP #43: CoffeeLakeFspBinPkg: Hangs w/ coreboot since update to FSP 7.0.68.40

n-huber, 2019-12-12:
Since commit f3ecfc4 (Coffee Lake FSP 7.0.68.40) we are unable to boot our CoffeeLake-S and -H systems. FSP-M simply hangs, no error reported.

Was this build tested? If so, with what hardware and bootloader combinations?

jackpot51, 2019-12-12:
@n-huber try turning off the FSP sharing coreboot stack

nate-desimone, 2019-12-12:
Hi @n-huber,

We recently ran into while working on TianoCore open source firmware actually. It fix it, we had to increase the size of the bootloader stack. Please see Prince's patch: https://edk2.groups.io/g/devel/message/52138

I think the 256KB is probably overkill, but it illustrates how to resolve the issue.

n-huber, 2019-12-13:
@jackpot51 thanks, but it's disabled by default. Seems to be part of the problem, though.

@nate-desimone Did you just switch to FSP 2.1 without notice??? Please update the documentation in case. I guess we can perform that switch for upstream coreboot. But what does Intel expect customers to do? Such a big switch late in a product cycle, plus all its legal concerns (remember coreboot is a GPL project, and Intel refuses to research compatibility).

n-huber, 2019-12-13:
I can confirm that it's working with increased coreboot stack and decreased heap size (config FSP_USES_CB_STACK). Please keep this issue open until _all_ related documentation is fixed!

nate-desimone, 2019-12-13:
@n-huber, no we did not switch CFL to FSP 2.1. We did implement the single stack feature from 2.1 in later versions of CFL FSP, but none of the bigger features like dispatch mode were ported.

nate-desimone, 2020-08-28:
@n-huber, I have updated the Coffee Lake FSP Integration Guide as of eb25f19 to explicitly document this. Note the following except from section 3.6.2:

> Starting with version 7.0.65.50 and later, this FSP will run on top of the stack provided by the bootloader instead of establishing a separate stack. This feature was adopted from the v2.1 FSP specification, however none of the other features described in v2.1 are implemented.

closed 2020-08-28 by nate-desimone