intel/FSP #45: KabyLake vbt file triggers Linux error message

wvervoorn, 2019-12-17:
The vbt files included in FSP 3.7.1 and later use a config size of 39 bytes. The last byte of this structure is not used according to the BSF.
This causes Linux to report : [drm:intel_bios_init] *ERROR* Unexpected child device config size 39 (expected 38 for VBT version 221).
At this point in time there is no functional problem with this but it would be better to either synchronize the Linux driver with the newer VBT so the 39 bytes are actually supported or report 38 instead of 39 bytes.

nate-desimone, 2020-01-07:
Hi @wvervoorn 

I'm not quite sure why the graphics team decided to add that extra byte. It is in the official VBT release that we got from them. Could you please open an issue with the Intel DRM team?

https://gitlab.freedesktop.org/drm/intel/issues

nate-desimone, 2020-03-06:
Hi @wvervoorn, did you have a chance to file the DRM issue?

nate-desimone, 2020-08-28:
Closing due to lack of response.

closed 2020-08-28 by nate-desimone