intel/FSP #47: MP_SERVICES pointer is optional

Th3Fanbus, 2020-01-08:
Hi,

As discussed [on review.coreboot.org](https://review.coreboot.org/36328), looks like the MP_SERVICES pointer is not a requirement:

> 1. The MP_SERVICES pointer is completely optional. You can set it to NULL and FSP will use its built-in MP implementation, which negates this whole argument. I am aware of several platforms that use coreboot with this set to NULL. It is the user's choice to configure coreboot in this way, not Intel's.

It would be nice to enhance the documentation with this additional information.

Thanks in advance,

Angel

nate-desimone, 2020-01-08:
Hi @Th3Fanbus - 100% agreed, I'll work on updating the relevant integration guides.

nate-desimone, 2020-08-28:
@Th3Fanbus, I've added some description text on CpuMpPpi to section 3.6.4 of the Coffee Lake FSP Integration Guide in eb25f19. Please take a look. I'm working on getting that added to the other Integration Guides now.

Th3Fanbus, 2020-08-30:
@nate-desimone I've checked the newly-added section and it looks good. Thank you!

nate-desimone, 2024-02-28:
Thanks for the feedback. Closing as it seems the issue is resolved.

closed 2024-02-28 by nate-desimone