intel/FSP #52: FSP DRAM configuration for memory with only single channel

shehamb, 2020-07-06:
Hello, 
We're using the custom hardware (let's call it BRC) based on the Apollo lake's Leaf hill crb. 

BRC -> 4 x 16 gib = 8 GiB of memory(4 dual-rank chips organized as 512 M by 32 bits.). This configuration i am able to boot up. 
 
Now, we have new variant of this board (just the memory downsized) with 1 x 8 gib = 1 GiB of LPDDR4 memory (1 single-rank chip organized as 256 M by 32 bits.).
(the other 3 chips were not populated).

(Note: Micron memory is being used : MT53E256M32D2DS-053 in both cases).
 
This is my DRAM config for these two boards:
8 GiB of memory(4 dual-rank chips organized as 512 M by 32 bits):
Channeln (n = 0, 1, 2, 3):
RankEnable             :0x1
DeviceWidth            :0x1
DramDensity            :0x4
Option                 :0x3
OdtConfig              :0x0
TristateClk1           :0x0
Mode2N                 :0x0
OdtLevels              :0x0
 
1 GiB of memory(1 single-rank chip organized as 256 M by 32 bits):
Channel0:
RankEnable             :0x1
DeviceWidth            :0x1
DramDensity            :0x2
Option                 :0x3
OdtConfig              :0x0
TristateClk1           :0x0
Mode2N                 :0x0
OdtLevels              :0x0
 
Channeln (n = 1, 2, 3):
RankEnable             :0x0
 
With this configuration, SBL is not getting past fspmemoryinit(). I am using the FSP debug library from Intel. From that, i am not quite sure what's going wrong. Does anyone of you have experience of using only single channel memory configuration?
 
I am enclosing the beyond compare snapshot between these two versions (working 4-channel board and single channel board). 
Also enclosing the console log for this single channel board. Any leads on this will be really helpful. 
[brc3-1gb-fsp-debug-only-RankEnable-disabled.txt](https://github.com/intel/FSP/files/4877081/brc3-1gb-fsp-debug-only-RankEnable-disabled.txt)

[image]

 
Regards, 
Mahesh.

mauricema, 2020-07-06:
Hi, Mahesh,

Maybe you already know that the memory configurations are highly tied to the board design.  And the board design needs to follow the APL platform design guide.  In your case, you used single rank device on single channel (assuming you are using LPDDR3 or LPDDR4),  I am not sure if it is currently supported or not.  All Intel's CRB board designs have more than 1 channel.  

If you look into some old APL EDS  volume 1 doc, you will see it listed supported memory configurations for APL.  It stated two channels are supported for "populating CH00 and CH01" and "populating CH10 and CH11".  And it also clearly stated "Single 32 bit channel" is not supported.   The latest EDS has removed that and stating it can support LPDDR4 memory configuration on CH0 (CH0_CQB) only.   Have your board team consulted Intel representative on this board design to ensure it follows the platform design guide ?  They might be able to share some of the info on how to provide correct memory configurations as well.

shehamb, 2020-07-07:
Hi Maurice,

Thank you for your reply. I think i know what the problem is. As you've
pointed out, per the EDS, mapping between channels and DQ is:
[image: image.png]
Ch-0 - CH0_DQB
Ch-1 - CH0_DQA
Ch-2 - CH1_DQB
Ch-3 - CH1_DQA

So, for a single channel, chip should be populated at CH0-DQB. But our
hardware engineer told me that it's populated at CH0-DQA. Since i am
accessing the board remotely (and hardware engineer is on leave this week),
i don't have any way to verify this immediately. But i am pretty sure this
is the problem.

I shall keep this forum posted once i get clarity on this.

Regards,
Mahesh.

sivabmc, 2021-03-29:
Any updates to the issue ? whether Intel FSP supports Single Rank configuration?

shehamb, 2021-03-30:
Yes. FSP does support single rank configuration.

This is the configuration we're using on our custom board (based on Leaf
Hill CRB):

Channel*0*:

RankEnable             :0x1

DeviceWidth            :0x1

DramDensity            :0x4

Option                 :0x3

OdtConfig              :0x0

TristateClk1           :0x0

Mode2N                 :0x0

OdtLevels              :0x0

Configured as 1 Rank (Rank 0 only) where each rank has a DramDensity (bit
density) of 16 Gib.

Hope this helps.

Regards,
Mahesh.