intel/FSP #58: Intermittent failure on APL platform with message "HeciInitialize HECIMBAR 0XD3708000"

shehamb, 2020-10-20:
Hello, 

We are experiencing significant manufacturing fallout (poor yield) on our custom design which is based on Apollo Lake (E3930). Some of the units boot-up quickly and reliably. Other units boot correctly immediately after power-up, but fail to boot on subsequent hardware or software resets.

We have adapted the Slim Bootloader BIOS to our product and built it with the Debug version of the FSP library MR6 ( i received this debug library from Intel rep).

When a unit fails to boot, it seems to pause at the following message in the boot log: 
 
HeciInitialize HECIMBAR 0XD3708000
 
Occasionally, after the unit pauses at this message, it automatically reboots, proceeds past this message on its second attempt, and successfully loads the operating system. In other situations, the self-reboot does not succeed and the unit remains stuck, displaying the second copy of the message indefinitely.

I had the following questions to Intel rep:

1.   What is the FSP library trying to accomplish when it displays this message "HeciInitialize HECIMBAR 0XD3708000"?
2.   What hardware or software failures would prevent this boot step from succeeding?
3.   Why does the Apollo Lake processor (or FSP library) reboot itself when this step fails the first time?
4.   After a self-reboot, why would a second attempt at executing the same code succeed? 
5.   Why would the FSP library stop self-reboot attempts? Does it count these attempts?
 
But Intel rep has come back to us saying that:
"According to Apollo Lake FSP Release Note, in chapter 8.0 "Hardware and Software Compatibility" stated that Coreboot is the supported bootloader. Besides that, based on IOTG Platform Boot Solution roadmap, Intel FSP with Slim Bootloader is not supporting Apollo Lake while Intel FSP with coreboot is supporting in Apollo Lake and you may refer to IOTG NDA 5Q Platform Roadmap for more details."

So, could someone please provide any leads on this issue?

Regards, 
Mahesh.

swong23, 2020-10-20:
Hi Mahesh,
     Yes, basically your Intel rep was right.
As this discussion involved IOTG NDA 5Q Platform Roadmap, we avoid further elaboration at this public portal here.

If your company is an Intel NDA customer, kindly file an IPS# ticket and more details about the road-map and support matrix could be explained there.

Thank you!

rprangar, 2020-10-23:
Hi Mahesh

As mentioned earlier, we can discuss any SBL related issues here. For FSP issues, you can follow up through the FSP channels.

Can we close this issue?

Thanks.

nate-desimone, 2020-12-17:
@shehamb, I'd recommend emailing @mauricema and @rprangar, they are probably help you with SBL issues.