intel/FSP #61: FSP 1.1 and later clarification

PQURETechnology, 2020-12-28:
Hi,

I have been looking for some information required for FSP 1.1 and later.

As FSP API has been split up to more calls, it is unknown using the new APIs wether TSEG, GFX buffers etc. has been initialised by FSP or not.

It would be natural to make the conclusion that they are initialised by the memory API function. It is important from a boot loader perspective to know when it initialised and the order of the stolen memory regions.

From some google source code for Skylake it looks like stolen memory is subtracted in the following order: BDSM, BGSM and TSEG.

The questions are:

1. Is the stolen memory regions above initialised in the memory init or silicon init functions?
2. In what order ?

A clarification would be greatly appreciated.

nate-desimone, 2024-02-28:
The number of stolen memory regions, their purpose, and their ordering can vary between Intel processor designs. The FSP specification does not require any specific behavior. Describing the behavior of stolen memory regions is one of the purposes of the FSP integration guide.

For example, here is the memory map for Skylake & Kabylake:

[image]

Hope this helps.

closed 2024-02-28 by nate-desimone

PQURETechnology, 2024-02-28:
FSP is unfortunately depending on several UEFI structures lik HOBs. So it's not an independent firmware initialising the DRAM and the silicon. So in many cases some boot loaders or BIOSes may not be able to use it. I integrated the 1.1 FSP in my BIOS, implementing the UEFI HOBs and all. It hang in the silicon init part. Later when implementing the BIOS on the same architecture without FSP, I found that some parts in the hardware was depending on what the BIOS has done before calling the FSP services. In order to be a real black box, the FSP should clearly define all outside dependencies. It should also define what parts of the silicon that should be initialised by the BIOS or boot loader. So in my point of view, the FSP should be totally independent of UEFI and clearly specify what's the BIOS/BootLoader job and whats the FSP job. It should also specify the memory regions and the structures used for calling the services should be proprietary for the FSP. It would then be a totally independent black box.