intel/FSP #63: TigerLake FSP problems with Tiger Lake UP3 RVP

miczyg1, 2021-02-11:
Tiger Lake FSP has problems with booting Tiger Lake UP3 RVP (DDR4 SODIMM design):

Tried MinPlatform with recently published TigerlakeOpenBoard and I get the following results using GCC5 compiler:
1. Building for FSP API mode:
- Memory Init fails quickly due to HDA GPIO configuration error (disabling HDA workarounds the problem)
- If the above is resolved then the FSP is stuck at `IgdIommuOverride` (at least this is the last log on the serial port with debug FSP binary)
2. Building for FSP wrapper mode:
- The platform resets after entering `PeiFspSaPolicyInitPreMem`. After the reset the last log is also an entrance to `PeiFspSaPolicyInitPreMem` but this time the platform hangs.

I experienced similar HDA problems with coreboot. Although I can get past FSP-M, the platform resets in CPU init routine during MCE registers clearing (https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/MW44TIEMFMVDWPVCAFPE2QUFXXGZYYAX/)
The errors may be related to the fact that port is done for LPDDR4 variant and I am testing DDR4 SODIMM.

@pietrushnic

pietrushnic, 2021-02-11:
@miczyg1 thank you for reporting that.

nate-desimone, 2021-02-22:
@miczyg1, the TigerlakeOpenBoardPkg GCC build errors were recently resolved: https://github.com/tianocore/edk2-platforms/compare/41b54dbb68036743768c53de9586f68a6d2c986f...9fb5174e0a164f83fc708c5941de6085e9cbb1c4

With regard to why you are seeing boot issues... I believe the problem is that TigerlakeOpenBoardPkg is currently only coded to support the Tiger Lake UP3 RVP with LPDDR4 and does not have the GPIO, SPD, or Audio Verb Table to support the DDR4 SODIMM RVP. I saw you filed a bugzilla as well (https://bugzilla.tianocore.org/show_bug.cgi?id=3219), thank you!

Do you happen to have a LPDDR4 RVP that you could try?

miczyg1, 2021-02-22:
@nate-desimone thank you so much.

Unfortunately, I do not have the LPDDR4 variant and the chances I will get one are very low.

nate-desimone, 2021-02-24:
@miczyg1, sorry I was incorrect in my last post. Looks like the DDR4 SODIMM RVP was the primary development vehicle for TigerlakeOpenBoardPkg actually. It looks like the problem is that your RVP currently has a A0 stepping SoC installed, which has been unsupported for several months now.

Please work with your Intel account representative to have your RVP upgraded to the newest Tiger Lake stepping. This should resolve your issues with both FSP and TigerlakeOpenBoardPkg.

miczyg1, 2021-02-24:
Thank you @nate-desimone . I will contact my FAE and try to persuade him to send us newer TGL stepping processor and link to your comments here and Bugzilla.

nate-desimone, 2021-02-24:
@miczyg1, your RVP may need some reworks to support the new stepping. Depending on the current rework state of your board, you may need to send it in for rework by an Intel technician.

pietrushnic, 2021-02-24:
@nate-desimone I appreciate Intel's policy about borrowing RVP - it is definitely superior to other vendors. Unfortunately, I cannot understand the use case for Open Source Firmware Vendor (OSFV) for stepping that is not supported in Open Source Firmware projects. It would be great to have some mapping that board goes to OSFV and some version simply does not make sense in that context. It would save some resources for everyone.

nate-desimone, 2021-02-25:
@pietrushnic, Intel only supports "for revenue" production CPUs on a long term basis. Engineering samples or quality samples like what you have right now are not supported for very long. I think you have the right hardware... you just need an upgrade to the newest stepping.

pietrushnic, 2021-02-25:
@nate-desimone we will try and hopefully can confirm all problems are resolved when correct stepping used.