intel/FSP #64: FspSiliconInit() requests reset under what circumstances?

shehamb, 2021-03-09:
Hello, 

we're are using FSP 2.0 library and are using Intel's Slimbootloader. Ours is customized hardware based on Intel's Leaf Hill CRB. 
We are trying to characterize the intermittent reboot issue. So for that we need to know more about:

it seems when FspSiliconInit() is called from the BIOS, FSP library may return the status either of this:
///
/// FSP Reset Status code
/// These are defined in FSP EAS v2.0 section 11.2.2 - OEM Status Code
/// @{
#define FSP_STATUS_RESET_REQUIRED_COLD         0x40000001
#define FSP_STATUS_RESET_REQUIRED_WARM         0x40000002
#define FSP_STATUS_RESET_REQUIRED_3            0x40000003
#define FSP_STATUS_RESET_REQUIRED_4            0x40000004
#define FSP_STATUS_RESET_REQUIRED_5            0x40000005
#define FSP_STATUS_RESET_REQUIRED_6            0x40000006
#define FSP_STATUS_RESET_REQUIRED_7            0x40000007
#define FSP_STATUS_RESET_REQUIRED_8            0x40000008
/// @}

Could we know under what conditions FSP will request reset?

Regards, 
Mahesh.

nate-desimone, 2021-03-09:
Hi,

On the Apollo Lake platform, when the NotifyPhase API is called for the PostPciBusEnumeration notification. The FSP may return FSP_STATUS_RESET_REQUIRED_COLD or FSP_STATUS_RESET_REQUIRED_5 if the PMC needs a reset to enable or disable south cluster devices. On the Apollo Lake platform, FSP_STATUS_RESET_REQUIRED_5 is returned when the FSP requests a global reset.

The PMC requires a reset to power gate some south cluster devices. For example, PCIe root ports and SATA ports. If any of these south cluster devices undergoes a state change since the last reset, a second reset may be required for PMC to adjust the device power state. This can also happen if the PMC losses power, which results in the previous state being lost.

nate-desimone, 2021-03-09:
I should also mention that this PMC behavior is somewhat unique to Apollo Lake.

nate-desimone, 2021-04-22:
Hope this answers your question, I'm closing this one for now. If you need more data go ahead and open a new issue.

closed 2021-04-22 by nate-desimone