intel/FSP #66: Booting IceLake server with EDK2 BIOS

jdhanasekar, 2021-04-21:
Hi Intel FSP Team,

I am having Ice Lake based server. I want to boot my server platform with EDK2 BIOS.  
Is it possible to boot the Ice Lake based server with this ```IceLakeBinPkg```  in EDK2 source code?

nate-desimone, 2021-04-22:
Hi @jdhanasekar, the `IcelakeFspBinPkg` is for the client version of Ice Lake, not the server one. We are working on releasing a `WhitleyFspBinPkg` along with a `WhitleyOpenBoardPkg` on [edk2-platforms](https://github.com/tianocore/edk2-platforms/tree/master/Platform/Intel) that will follow the [MinPlatform Architecture](https://tianocore-docs.github.io/edk2-MinimumPlatformSpecification/draft/) and boot Ice Lake based servers.

jdhanasekar, 2021-04-22:
Hi @nate-desimone, Thanks for the info. I am having ```coyote pass server``` and will ``WhitleyOpenBoardPkg`` support coyote pass server platform or I have to port platform changes accordingly?

pietrushnic, 2021-04-22:
@nate-desimone is there any way to obtain FSP release roadmap?

nate-desimone, 2021-04-22:
@jdhanasekar Our initial release will support Wilson City. We didn't any explicit plans to add support for Coyote Pass but it would not be very difficult to do so. If you are interested in helping with that effort after we get `WhitleyOpenBoardPkg` that would be very appreciated.

@pietrushnic, we do not have a roadmap.

jdhanasekar, 2021-04-23:
@jdhanasekar . I am interested in helping to port changes for ``coyote pass`` in ``WhitleyOpenBoardPkg`` once it is released.
Please inform me once WhitleyOpenBoardPkg is released.

closed 2021-05-21 by jdhanasekar

paulmenzel, 2021-05-21:
Why did you close the issue? Has WhitleyFspBinPkg been released?

jdhanasekar, 2021-05-22:
WhitelyFspBinPkg is not release yet.  I thought I got the required info. So, I am closed the issue.  Do You want me to keep the issue open till they release WhitelyFspBinPkg for  Wilson City CRB ?