intel/FSP #68: Tiger Lake FSP has problem with DDI TCP HDMI port

PennZheng, 2021-07-08:
Hi,

I'm working on a new designed Tiger Lake-U board.
The Slim Bootloader and UEFI Payload I used was download on April 2021.

I encounter a problem that the DDI TCP1 HDMI port cannot work in SBL with FSP provided GOP driver.
The hardware design may be no problem since the DDI TCP1 HDMI port can function properly in Windows 10 with Intel Graphics Driver.

Do you have any idea on this problem?

Thanks

swong23, 2021-07-08:
**Clarification needed:**
-> What does it mean by: "The hardware design may be no problem since the DDI TCP1 HDMI port can function properly in Windows 10 with Intel Graphics Driver."?

- If it cannot work in SBL with FSP provided GOP driver,
- How it able to function properly in Windows 10 with Intel Graphics Driver?

Thanks!

PennZheng, 2021-07-08:
I mean the DDI TCP1 HDMI port cannot work before UEFI Payload boot Windows 10.
- The monitor does not show the Slim Bootloader logo when SBL is running.
- It says in the SBL debug log: "Failed to get GFX HOB from FSP".

Thanks

PennZheng, 2021-07-09:
Hi,

Additional information:

The DDI B HDMI Port functions properly in Windows 10 with Intel Graphics Driver and also functions properly in SBL and UEFI Payload.
The DDI TCP1HDMI Port functions properly in Windows 10 with Intel Graphics Driver but not works in SBL and UEFI Payload.

The full debug logs during boot are attached here:
[G3_to_S0_monitor_connected_to_DDI_B_HDMI_Port.log](https://github.com/intel/FSP/files/6788564/G3_to_S0_monitor_connected_to_DDI_B_HDMI_Port.log)
[G3_to_S0_monitor_connected_to_DDI_TCP1_HDMI_Port.log](https://github.com/intel/FSP/files/6788565/G3_to_S0_monitor_connected_to_DDI_TCP1_HDMI_Port.log)

Thanks.

closed 2021-09-03 by PennZheng

Advgcipc, 2022-01-19:
I also meet the same issue. Did possible share how to fix that?
Thanks!
debug log:
FspGfxHob is not available
Boot-time Cpu DTS: 0x12C
Failed to get GFX HOB from FSP