intel/FSP #69: Apollo Lake MR9 FSP will set SMI Lock no matter what value I passed from bootloader

seattlezhu, 2021-07-16:
Hi,

I found no matter what value of LockDownGlobalSmi I passed form bootload,  FSP would set SMI Lock after Notify EnumInitPhaseReadyToBoot. Is this a problem or out of some particular design? How would I supposed to clear SMI Lock？

Thanks!

nuramir1, 2021-08-06:
Hi @seattlezhu,

Thank you for reporting this to us.
Could you clarify more on how do you get this failure on your system? What test and settings have you done that leading to this error?

Thank you.

nuramir1, 2021-08-17:
Hi @seattlezhu,

Is there any update on the request above? It'll be great if you could clarify more so that we can proceed to debug this issue.

Thank you.

seattlezhu, 2021-09-03:
Hi @nuramir1 ,

Sorry for the late reply .
I found this issue when I want to use SMI in UEFI Payload. To use SMI, I need to set Global SMI Enable bit in ABASE. But this bit cannot be changed if SMI Lock bit is set. So I set LockDownGlobalSmi to disable in Slimboot, but found out FSP would set SMI Lock after Notify EnumInitPhaseReadyToBoot no matter what value I passed. This could  be observed from debug print of SMI Lock register value.  

Thank you.

nuramir1, 2021-09-08:
Hi @seattlezhu,

Noted that this issue is being tracked and I will get back to you soon if there is any update on this issue.

Thank you.

nuramir1, 2021-09-14:
Hi @seattlezhu ,

Can you share your debug log for the failing system? We would like to check on the SMI Lock debug print. Have you tried with previous MR6 FSP whether the issue is happening or not?

Thank you,
Amirah

seattlezhu, 2021-09-18:
Hi @nuramir1 ,

I have tried with MR6 FSP and result to the same. The debug log is pretty easy just print the value of SMI Lock register at dirfferent InitPhase in Stage2 BoardInit function if you use Slimboot as the host bootloader.

Thank you.

nuramir1, 2022-02-15:
Hi @seattlezhu,

Sorry for the late reply as we're working on the latest FSP releases.
Can you please try with MR10 FSP and see whether the issue still occurs or not?

Thank you,
Amirah