intel/FSP #78: GLK - FSP M configuration

Sean-StarLabs, 2022-01-10:
I'm trying to port a GLK based board, and despite configuring FSP parameters according to the datasheet, it won't get past FSPM.

Whilst the debug binary provides a lot of info, I can't find any info on how to use this information to fix the configurations.

Would you be able to point me in the right direction, either how to diagnose the error or what the problem is?

```
============= FSP Spec v2.0 Header Revision v3 ($GLKFSP$ v2.2.0.0) =============
Fsp BootFirmwareVolumeBase - 0xFEF40000
Fsp BootFirmwareVolumeSize - 0x66000
Fsp TemporaryRamBase       - 0xFEFD6000
Fsp TemporaryRamSize       - 0x2A000
Fsp PeiTemporaryRamBase    - 0xFEFD6000
Fsp PeiTemporaryRamSize    - 0x1B4CC
Fsp StackBase              - 0xFEFF14CC
Fsp StackSize              - 0xEB34
GPIO pad is not owned by host! skip it. Ownership: 03
GPIO pad is not owned by host! skip it. Ownership: 03
Install PPI: 8C8CE578-8A3D-4F1C-9935-896185C32DD3
Install PPI: 5473C07A-3DCB-4DCA-BD6F-1E9689E7349A
The 0th FV start address is 0x000FEF40000, size is 0x00066000, handle is 0x0
Install PPI: B9E0ABFE-5979-4914-977F-6DEE78C278A6
Install PPI: DBE23AA9-A345-4B97-85B6-B226F1617389
Loading PEIM at 0x000FEF4EC90 EntryPoint=0x000FEF4EFE5 PcdPeim.efi
Install PPI: 06E81C58-4AD7-44BC-8390-F10265F72480
Install PPI: 01F34D25-4DE2-23AD-3FF3-36353FF323F1
Loading PEIM at 0x000FEF503E4 EntryPoint=0x000FEF50C9C FspInitPreMem.efi
Install PPI: 820695B0-C2F2-4EC7-A252-DB0DBEE838A3
Install PPI: CFE5EC91-31ED-47E9-BE7D-9CCB59134B71
Install PPI: F8D5438E-26E1-481D-B63C-30D6EFF4A420
Install PPI: CBD86677-362F-4C04-9459-A741326E05CF
Install PPI: 563F8EDE-1FA5-45A2-BE23-B0B6A07DE239
Loading PEIM at 0x000FEF54274 EntryPoint=0x000FEF56A22 SiInitPreMem.efi
GPIO pad is not owned by host! skip it. Ownership: 03
GPIO pad is not owned by host! skip it. Ownership: 03
SiInitPrePolicy() Start
ScInitPrePolicy() - Start
POSTCODE=<0000DB00>
ScInitPreMem() - Start
POSTCODE=<0000DB02>
ScEarlyInit() - Start
ScSmbusInit() Start
ScSmbusInit() End
ConfigureLpcOnEarlyPei() 
ScEarlyInit() - End
POSTCODE=<0000DB7F>
ScInitPreMem() - End
ScInitPrePolicy() - End
POSTCODE=<0000DB10>
ScConfigurePciePowerSequence ()
Clearing BLKDQDA and BLKPLLEN
Clearing BLKDQDA and BLKPLLEN
Clearing BLKDQDA and BLKPLLEN
Clearing BLKDQDA and BLKPLLEN
Clearing BLKDQDA and BLKPLLEN
Clearing BLKDQDA and BLKPLLEN
Pre-Mem North Cluster Entry
POSTCODE=<0000DA00>
PeiNpkInit() - Start
NpkReserveHob->NpkPreMemConfig.TraceHubEn = 0x0
NpkReserveHob->NpkPreMemConfig.TraceHubVrcTapEn = 0x0
NpkReserveHob->NpkPreMemConfig.FwTraceEn = 0x1
NpkReserveHob->NpkPreMemConfig.RecoverDump = 0x0
NpkReserveHob->NpkPreMemConfig.FwTraceDestination = 0x4
NpkReserveHob->NpkPreMemConfig.Msc0Size = 0x0
NpkReserveHob->NpkPreMemConfig.Msc0Wrap = 0x1
NpkReserveHob->NpkPreMemConfig.Msc1Size = 0x0
NpkReserveHob->NpkPreMemConfig.Msc1Wrap = 0x1
NpkReserveHob->NpkPreMemConfig.PtiMode = 0x1
NpkReserveHob->NpkPreMemConfig.PtiTraining = 0x0
[... 2407 lines cut ...]
  CH03    0245:-   
ECT: S05000000, E33003300
  CH03    0215:-   
ECT: S05000000, E33003300
  CH03    0185:-   
ECT: S35000000, E33003300
  CH03    0155:-   
ECT: S30000000, E33003300
  CH03    0125:-   
ERROR: Highest Value found is a 0
Final solution Rank 0 = 0, 0, Final control is 123, Final command is 125
CP 56
Won't run on disabled channels
 Lp4CBTExitAllRanks
Won't run on disabled channels
 Lp4CBTExitAllRanks
Won't run on disabled channels
 Lp4CBTExitAllRanks
Won't run on disabled channels
 Lp4CBTExitAllRanks
CP B2
 CompleteJedecInit
DRAM Resetting
Set Boot Freq:3 Speed set to:3
```

Sean-StarLabs, 2022-01-11:
No, they haven't

nate-desimone, 2022-01-11:
Hi @Sean-StarLabs

First of all, it is inappropriate to questions and full debug logs on FSP binaries that are not public. Please work issues like this through your Intel account representative in the future.

Second, here is the important error from your output:

`
ERROR: Highest Value found is a 0
`

That means memory errors are encountered in 100% of all possible operating points. There is clearly something very wrong with the UPD parameters you are passing in.

Given that this is not actually a bug in the FSP, I am closing this issue. This forum is for reporting FSP bugs, it is not a general FSP support help-desk.

closed 2022-01-11 by nate-desimone

Sean-StarLabs, 2022-01-11:
I didn't know the account reps could assist with anything technical, I'll reach out. Thanks :)

nate-desimone, 2022-01-11:
Usually not directly, but they should be able to forward your query to the tech marketing engineers covering Gemini Lake :)

miczyg1, 2022-01-12:
@Sean-StarLabs you might want to try a newer version of FSP. There is v2.2.1.3 available for sure.