intel/FSP #83: Alder Lake S IOT FSP package has missing headers

miczyg1, 2022-04-13:
The Include directory is missing MemInfoHob.h and FirmwareVersionInfoHob.h files.

nate-desimone, 2022-06-29:
I included them in the client release:

https://github.com/intel/FSP/blob/master/AlderLakeFspBinPkg/Client/AlderLakeS/Include/MemInfoHob.h
https://github.com/intel/FSP/blob/master/AlderLakeFspBinPkg/Client/AlderLakeS/Include/FirmwareVersionInfoHob.h

As long as the IoT owners approve I can copy those over to the IoT folder as well.

miczyg1, 2022-06-29:
@nate-desimone WOW, the CLient FSP is already here! That's great.

I also have contacted our FAE to include these files, but apparently MemInfoHob.h is confidential, which is confusing to me. Especially when Slimbootloader has the file present on public repo: https://github.com/slimbootloader/slimbootloader/blob/master/Silicon/AlderlakePkg/Include/MemInfoHob.h  (although I have no means to confirm it is sourced from IOT FSP).

miczyg1, 2023-07-06:
Both files are already public:
https://github.com/intel/FSP/blob/master/AlderLakeFspBinPkg/IoT/AlderLakeS/Include/MemInfoHob.h
https://github.com/intel/FSP/blob/master/AlderLakeFspBinPkg/IoT/AlderLakeS/Include/FirmwareVersionInfoHob.h
Closing

closed 2023-07-06 by miczyg1