intel/FSP #95: Sapphire Rapids FSP

pp3345, 2023-10-09:
Does Intel plan to release a coreboot-compatible FSP for Sapphire Rapids CPUs / Eagle Stream platform, and if so, is there an estimate available on when the FSP will be published?

65a, 2023-11-03:
Hoping Intel wants to sell more chips and responds! Seeing the public contributions to coreboot from Intel and ODMs for SPR-SP is great, but none of it is usable months later, because it is not possible to get the FSP binaries that were used. I have a COTS board that could probably be added to the tree today if I could actually build coreboot with the necessary FSP blobs.

LeanSheng, 2023-12-22:
hi guys, quick update: the SPR (Sapphire Rapids) FSP is here :D
https://github.com/intel/FSP/tree/master/EagleStreamFspBinPkg
Feel free to reach me out if need any help on integrating them for coreboot usage ;)
sheng.tan@9elements.com

nate-desimone, 2024-01-02:
Fixed in 848c499

closed 2024-01-02 by nate-desimone