intel/FSP #97: Building FSP for Sapphire Rapids

abanejad, 2023-12-10:
Hello,
I am trying to port edk2 on a Sapphire Rapids but the main problem i have is that the FSP binaries are not available so I was reading this paper "A Tour Beyond BIOS Creating the Intel® Firmware Support Package Version 1.1 with the EFI Developer Kit II" and I guess implied the idea that a FSP binary could be created with edk2's IntelFsp2Pkg and providing the associated silicon and platform pakages. So I was wondering would this actually be possible no matter how hard it would be? or do I have to have some proprietary resources?
Thanks

LeanSheng, 2023-12-22:
Hi @abanejad ,
No it is not possible to create the FSP from scratch for specific platform, as FSP does a lot of architecture/ silicon specific initialisation. Good news is Intel just updated the Sapphire Rapids FSP here!
https://github.com/intel/FSP/tree/master/EagleStreamFspBinPkg
And it should be not hard to power up the platform as coreboot already has stable Sapphire Rapids support:
https://github.com/coreboot/coreboot/tree/main/src/soc/intel/xeon_sp/spr
Feel free to reach out to me if you need any further help to integrate with coreboot :)
sheng.tan@9elements.com

abanejad, 2023-12-23:
Hi @LeanSheng,
thanks for the update. Now I can build my ifwi. Too bad I had many complications with building and executing coreboot firmwares in the past and I think I'll stick to edk2-platforms since I find the UEFI structure easier to comprehend since it has more docs even though it has a much more complex build system. Also I find coreboot's motherboard porting method very insufficient and confusing. It seems the coreboot community isn't as motivated as you might expect when it comes to presenting and documenting.

closed 2023-12-23 by abanejad