intel/FSP #99: BCT can't open Eagle Stream's bsf

abanejad, 2024-01-10:
Hello
I am trying to use the bct tool to patch the fsp.fd file. Unfortunately BCT isn't able to open the associated BSF file.
[image].

swong23, 2024-01-10:
Please try using the Config Editor tool -

Config Editor tool for FSP configuration
Link: https://github.com/tianocore/edk2/tree/master/IntelFsp2Pkg/Tools/ConfigEditor

Readme: https://github.com/tianocore/edk2/blob/master/IntelFsp2Pkg/Tools/UserManuals/ConfigEditorUserManual.md

abanejad, 2024-01-10:
@swong23 Thank you.
I don't know if you have checked this yourself or not. But unfortunately the config Editor Tool doesn't load the parameters of Eagle Streams BSF file either

LeanSheng, 2024-01-12:
usually, you can directly configure the UPD parameters directly from BIOS like coreboot or Slim bootloader, so BCT tool or config editor tool comes secondary. it is because the tool only changes the 'default' settings of FSP, if BIOS reference code like coreboot overwrites the UPD settings, then the default setting will be ignored.

memristor2, 2024-02-05:
Fixed in #100

paulmenzel, 2024-02-05:
The merge/pull request is still open, and I think it’s common practice to leave the corresponding issue open until the merge/pull request is accepted and merged/submitted. If you add a line

> Resolves: https://github.com/intel/FSP/issues/99

to the commit message of the commit fixing the issue, then GitHub is going to close the issue automatically, when the merge/pull request is merged/submitted.

memristor2, 2024-02-05:
@paulmenzel I simply did this to close it since it was not an issue for me anymore. But you're right Ill reopen it and let the contributers close it

nate-desimone, 2024-02-27:
Fixed in `e07f875`

closed 2024-02-27 by nate-desimone