intel/FSP #101: UP3 Tigerlake Based Circuit Board: CRC For DIMM Fails. So Fast Boot Never Occurs!

sshivsh, 2024-02-05:
[Link to Intel Support Forum](https://community.intel.com/t5/Embedded-Intel-Core-Processors/No-Boot-Time-Difference-First-And-Subsequent-Boots-After/td-p/1567432?emcs_t=S2h8ZW1haWx8YWNjZXB0X2FzX3NvbHV0aW9uX3JlbWluZGVyfExTNzlPSFQ4OExMWjE5fDE1Njc0OTJ8T1RIRVJTfGhL)

Hi. From the forum at the link above, we were redirected here. We have mentioned further details below:

Please check the attached log for the messages surrounding the text below.
[debug2.log] (https://github.com/intel/FSP/files/14163336/debug2.log)

```
MRC task -- Fast boot permitted -- Started.
Channel 0 Dimm 0 DimmCrc C97Eh, DimmSave->Crc C97Eh
Channel 0 Dimm 1 DimmCrc C97Eh, DimmSave->Crc 0h
Dimm has changed, cold boot required
MRC task Fast boot permitted -- FAILED, Status = 18h.
```

MRC Fast Boot is failing because of this. So the saved memory training results are never used after the 1st boot. Instead, cold boot
keeps occurring on the 2nd boot, 3rd boot etc.

Does someone know why the CRC error occurs? How can we prevent the error & enable Fast Boot?

Thanks!

Josh0041, 2024-02-07:
Good afternoon. I'm Josh and I was made aware of your issue. I need to confirm some of your board details:
1. What TGL-UP3 stepping are you using?
2. What reference board did you based your custom board? 
3. I saw from the logs that you are using LPDDR4 16GB Soldered/Memory Down. Can you give me the brand and other details?
4. I have attached my debug log from my reference board. The main difference other than the memory size is that there is a mismatch on the CRC on Channel 0 Dimm 1.

You can take a look and compare our debug logs. I will check further on what is the possible problem.

With best regards,
Josh

[putty_tglup3_20240208005856.log](https://github.com/intel/FSP/files/14191858/putty_tglup3_20240208005856.log)

sshivsh, 2024-02-07:
Hi Josh.

1. We use stepping 1.
2. We used the RVP reference board. LPDDR4 Memory routing is based on "609182_TigerLake_UP3_TGL_UP3_LPDDR4x_AEP_TDK_Rev1p0". Rest of the platform is based on "612304_TGL_UP3_LPDDR4x_RVP_TDK_Rev1p0"
3. The LPDDR part no. is MT53D1024M32D4 (from Micron).

Timing values can be configured through the **Spd_Lpddr4.bin** file in Slimbootloader. I've shared this file at the link below:
[Spd_Lpddr4.bin](https://drive.google.com/file/d/1TvBqKtjvDlR5rKzvYf1Vo9v272YTdb52/view?usp=sharing)

Thanks for your debug log. A question: How is the DIMM CRC calculated & for what purpose?

Thanks!

Josh0041, 2024-02-08:
Hi SShvish.

You are referring to the B1 Stepping, correct?

From what I know, the CRC16 is needed to detect a change in the DIMM module. Since your board is memory down, this is not possible.

The data from your log showed that a CRC is generated but the stored CRC from the previous Cold Boot is 0. That is the reason why it always does a Cold Boot.

It will take some time for me to root cause this, but I need to double confirm with you. Are you able to utilize or see the full 16GB main memory? In EFI Shell or OS?

With best regards,
Josh

sshivsh, 2024-02-08:
Hi Josh.

Here is some information from different lines in the debug log:
    
    PCH Stepping : B0

    TigerLake ULT: Stepping B0

    CpuStepping: 1h

Yes. We are able to see the full 16 GB of memory in the OS.

Thanks!

sshivsh, 2024-02-08:
Also, from a photo of the processor:

    CPU Spec Code: SRK0Y
    
    Stepping: B1

Thanks!

sshivsh, 2024-02-13:
Hi Josh, Were you able to progress towards root-cause? Thanks!

Josh0041, 2024-02-15:
Hi SShvish,

Good evening. Sorry, I just returned from the long holiday. This is still in progress. 

With best regards,
Josh

Josh0041, 2024-02-16:
Hi SShvish,

Good afternoon. Can you send an IPS ticket regarding this issue? I will take ownership of the ticket once it is posted.

With best regards,
Josh

sshivsh, 2024-02-16:
Hi Josh.

We don't have IPS privilege. How can we obtain it?

Thanks.

Josh0041, 2024-02-16:
Hi SShvish,

Normally, your Intel representative (PAE) will be the one to give you access to IPS as an NDA partner. He/She is the one who provided the reference documents you mentioned earlier. Are you able to contact this person?

I will reply on your original thread for someone who can assist on the IPS access.

Regarding the progress... I might have found a fix, but I am unsure yet as we have different memory configuration. 

With best regards,
Josh

Josh0041, 2024-02-17:
Hi SShvish,

Good morning. I have responded in your original thread. Please contact your local FAE regarding IPS access.

With best regards,
Josh

sshivsh, 2024-02-17:
Hi Josh. Sorry for the delay: Actually, my company works till 7 PM on weekdays, but not on Saturday & Sunday!

I noticed your post, thanks a lot. I will get back to you on the coming Monday!

sshivsh, 2024-03-13:
Hi Josh,

I got the ability to make a new IPS case (with the Tigerlake product) only now. Sorry for the delay!

The IPS Case Number is 00868720.

Hope you can help us now. Thanks!

Josh0041, 2024-03-14:
Hi Siddhartha,

Good morning. That is great! I just took ownership. Our succeeding communique will proceed from there.

With best regards,
Josh

swong23, 2025-02-17:
IPS Case Number is 00868720 closed: User to disable all the MEMORY_CFG_DATA.SpdDataSelXY1 instances in SBL.

closed 2025-02-17 by swong23