intel/FSP #104: Issues with EagleStream/SPR-SP FSP

65a, 2024-03-16:
I'm trying to port the Tyan S5652 board to Coreboot. Using a gpio dump from inteltool, I have the running GPIO settings for the board. I started with Archercity CRB as a template, but updated the IIO configs to match my board. I immediately ran into some problems:

1. #102 The headers are incomplete in the FSP git, and out-of-date in Coreboot
2. The FSP will not log debug logs regardless of UPD configuration (SerialIODebugEnable, 0x3f8 is the default, loglevel 0x3, etc)

I hacked around 1 by merging the few headers in this repo back into Coreboot's old FSP headers, this sort of seems to work and stops triggering asserts. I never solved 2.

At this point, I was able to boot the board past romstage, and FspMemoryInit succeeds. In ramstage, I noticed new problems:
1. Memory is detected but no dimms are listed
2. If I try to run a VGA option rom, the rom will execute, but loops reading some VGA registers (I did try to set the LegacyVgaSoc and LegacyVgaStack UPDs, no difference, but more docs would be great). I tried native execution and YABEL in various configurations, none of these worked, although I could share the yabel register access logs if that's useful.
3. If any* (see below) payload tries to boot or run option roms, the machine reboots.

I suspect this all might be fallout from #102, but as far as I can tell the current FSP does not work in Coreboot, and most of the work seems to be happening by Intel insiders with access to development FSPs, headers and docs. Am I doing something wrong here? 

The following payloads do not work at all:
1. Seabios
2. Linux bzImage
3. Grub2 (prints an Exception:14 Apic:0 error)

I **was** able to get to a u-boot flat binary prompt with that payload, but since it cannot see through PCI bridges (apparently), I cannot boot with that either. I did not try EDK2 because I couldn't get VGA working.

Could Intel please try to build Archercity with **only** public files and confirm it actually works? Is there something obvious I am missing?

65a, 2024-03-17:
Thanks to @johnnylinwiwynn I was able to boot SeaBIOS, Linux, and u-boot. I didn't try grub again. Going over the configs carefully, I changed to XAPIC mode, enabled VPD FMAP, and properly configured the CXL. This boots fine, however option ROMs hang and do not work. I suspect this is because the legacy 8254 timer is clock gated, as older FSPs had a special config to keep this on, and Option ROMs appear to require it.

closed 2024-03-17 by 65a

MiTACWilson, 2024-06-13:
Hi 65a,
I have the same problem when porting coreboot on our board. Some questions needs your help. 
Would you please share the detail configurations with me ?
1. How to enable XAPIC mode ? 
2. What keys and values are put in the VFD FMAP ?
3. I changed the IIO PCIe Subsystem Mode to CXL. Is it properly ?
4. What FSP binary and it's related include files you used ?
Thanks !!

65a, 2024-06-14:
It wasn't XAPIC, though in `make menuconfig` you should adjust that setting. Instead, in the security section of coreboot config, set it to clear memory on every boot. You can usually boot without VPD FMAP keys set, but digging in the code will show you the options. For CXL, I don't use it, configuring any value is fine: more important to ensure the IOUs are configured appropriately for PCIe devices. Use the FSPs associated with this repo, include files from coreboot + FSP repo. Good luck, if you are from MITAC, I did get my Tyan board booting, as well as a Gigabyte, reliably.

MiTACWilson, 2024-06-14:
Hi 65a,
Thanks your comments. 
Yes, I am from MiTAC. 
We got our board booting with the workaround on the link https://review.coreboot.org/c/coreboot/+/82705 yesterday.
The paging function of latest EGS FSP binary should be disabled to linux payload. 
Did you patch this workaround  into your coreboot source tree ?