intel/FSP #110: Unable to set XMP profiles with RaptorLakeFSP without bricking firmware

shakealeg, 2024-09-25:
This is being tested on Raptor Lake FSP from this repository, i9-14900K, 2x48GB 5600MHz Crucial Pro DDR5 with XMP 3.0 support.

I'm on the stage in our firmware building where I want to add XMP. I have attempted to add `SpdProfileSelected` to my FSPM configuration, but it does not boot. I saw some more manual configuration options in the `FspmUpd.h` file, but I'd like to hear something here.

When our firmware is flashed with this new config set, it does not boot. Memory does not train.
Do we require some more configuration options?

Following the comments on SpdProfileSelected:
```
Offset 0x01E1 - SPD Profile Selected
0:Default SPD Profile, 1:Custom Profile, 2:XMP Profile 1, 3:XMP Profile 2, 4:XMP Profile 3, 5:XMP User Profile 4, 6:XMP User Profile 5
**/
```
The Crucial Pro RAM only has 2 XMP 3.0 profiles:
XMP 3.0 Profile 1; 5600MT/s 46-45-45
XMP 3.0 Profile 2; 5200MT/s 42-42-42
We've tried setting `SpdProfileSelected` to either `2` (XMP Profile 1) or `3` (XMP Profile 2), yet no luck.

I would be grateful for some advice.

Thanks

shakealeg, 2024-09-25:
Feel free to ask for anything additional if needed.

shakealeg, 2024-10-11:
Real professional guys, waste of time. Will look into AMD.
Leaving this open so it can still be fixed.

nate-desimone, 2024-10-11:
@shakealeg - Overclocking is not guaranteed to work. It sounds like you are able to boot at the validated DDR frequencies, correct?

shakealeg, 2024-10-12:
> @shakealeg - Overclocking is not guaranteed to work. It sounds like you are able to boot at the validated DDR frequencies, correct?

Thank you for the response. I understand DDR5 overclocking is not guaranteed, but XMP with these sticks works on all competitor boards.
I'm not exactly sure what your question means, but the board I'm creating a firmware for has these tested frequencies:
```
Memory Support: 7000+(OC)/ 6800(OC)/ 6600(OC)/ 6400(OC)/ 6200(OC)/ 6000(OC)/ 5800(OC)/ 5600(JEDEC)/ 5400(JEDEC)/ 5200(JEDEC)/ 5000(JEDEC)/ 4800(JEDEC) MHz
Max. overclocking frequency:
• 1DPC 1R; 7000+ MHz
• 1DPC 2R; 6600+ MHz
• 2DPC 1R; 6400+ MHz
• 2DPC 2R; 5600+ MHz
```
I am able to boot at Intel default spec which is 4400. I also found that when I set `DdrFreqLimit` to `5600`, it starts running even higher at 4800.

nate-desimone, 2024-10-14:
On Raptor Lake, Intel officially supports the up to DDR5 5600 with JEDEC specified timings. Because you choose to run at 5600 and the memory was only to train up to 4800, it sounds like there is some marginality with one (or more) of the following components:

1. The CPU
2. The board
3. The DRAM

The fact that you are not able to train up to the fastest officially validated frequency, it makes sense that attempting to overclock results in a boot failure.

One possible explanation for this is that the manufacturer of the board in question has made changes to the memory training code that is not present in Intel's vanilla reference code and therefore would not be available in the form of an FSP. If you try using the stock firmware for the board does it train to 5600? If it trains to 5600, can you overclock with the stock firmware?

nate-desimone, 2024-11-11:
Any update? If you try using the stock firmware for the board, does it train to 5600? If it trains to 5600, can you overclock with the stock firmware?

shakealeg, 2024-11-11:
Sorry for late reply Nate, I thought I responded before. Must have forgot
to press send. Yes, on stock firmware, it trains to 5600 by default. And
yes, XMP works too.

nate-desimone, 2024-12-18:
Given that... the only thing that makes sense is that the board vendor has modified the memory training algorithms in some way. At Intel, we have noticed that board vendors do tend to do that.

Therefore, I think your best bet is to try to extract and use the FSP binary from your board vendor's firmware instead of trying to use the reference FSP binaries provided by Intel. Best of luck!

closed 2024-12-18 by nate-desimone