intel/FSP #113: Clarification of "binary form, without modification" in FSP License

nic3-14159, 2024-12-18:
As per the FSP License, "Redistribution and use in binary form, without modification" is permitted (provided the other conditions are met). Is this meant to target the individual FSP modules (UEFI FFS files), or does it apply to larger collections such as the split FVs or even the entire Fsp.fd binary as distributed by this repository?

Given that actions like splitting the Fsp.fd file into individual FVs, rebasing the FSP, and modifying the UPD parameters technically result in a modified binary if compared to the Fsp.fd binaries distributed in this repository, and that the resulting binaries from such actions seem to be redistributable in practice, it would seem that "binary form, without modification" is intended to apply at the level of the individual FSP modules (FFS files).

The motivation for this question is the ideas discussed in [1], which propose removing unused FSP modules from the Firmware Volumes in cases where there is equivalent code in a consumer such as coreboot or where certain functionality is not needed. The resulting FD or FVs would technically be "modified" in comparison to the original Fsp.fd binaries from this repository and the FVs contained in them, but the modules themselves remain unmodified.

[1] https://blog.osfw.foundation/breaking-the-boundary-a-way-to-create-your-own-fsp-binary/

nate-desimone, 2024-12-18:
An FSP binary modified in the manner described in the above article would not be redistributable under the FSP license in this repository. You can redistribute the tools and instructions described in that article, but not the resulting image.

closed 2024-12-18 by nate-desimone

LeanSheng, 2024-12-19:
> An FSP binary modified in the manner described in the above article would not be redistributable under the FSP license in this repository. You can redistribute the tools and instructions described in that article, but not the resulting image.

@nate-desimone are you 100% sure? Wouldn’t it be better to consult legal first?

and are you still in FSP/BIOS team? If not I am not sure if you could answer on behalf of them.

nic3-14159, 2024-12-19:
I guess what I'm actually after is a clearer definition of what the "binary" being licensed is. I brought up that article as an example, though it did make me think about the broader interpretation of the license. As you may know, the FSP license was previously changed to match the license of the Intel microcode repository [1]. Those microcode binaries are typically used as is, in their entirety, without manipulation, and thus the interpretation of the license is clear.

In the case of the FSP, there are several common actions (as I mentioned in my original question) that result in a binary that is technically different than the Fsp.fd files in this repo:

- Splitting the FSP into individual firmware volumes, which results in files that are subsets of Fsp.fd
- Rebasing the FSP, which changes addresses in the individual modules within the components, as well as the FSP header
- Changing the static UPD parameters, which modifies the UPD file

If "binary" is intended to mean Fsp.fd, then would that mean that firmware solutions such as a coreboot rom could not be distributed unless they contain the entirety of Fsp.fd as distributed in this repository, without any combination of splitting, rebasing, or changing UPD parameters? Splitting Fsp.fd, dropping FSP-T, and rebasing FSP-M and S are all common actions in coreboot.

Even if "binary" was intended to mean the components, allowing splitting the Fsp into components (Fsp.fd is simply a direct concatenation of the individual components from what I can tell) the above issue would still be present due to actions like rebasing and changing UPD parameters.

So if the actions described in that article result in a non redistributable binary, then I believe that either all of the actions I just mentioned result in non-redistributable binaries; or the license is not clear about what changes are permitted regarding redistribution and what are not, if binaries produced from those other actions are redistributable.

This has implications for the various coreboot distributions [2], which often distribute ROM images containing the FSP with the above mentioned actions (excluding what was described in that article) applied.

[1] https://www.phoronix.com/news/Intel-Better-FSP-License
[2] https://doc.coreboot.org/distributions.html

nate-desimone, 2024-12-19:
Intel's policy is to not offer legal advice. If you need more clarification on the license, the I suggest you consult with a lawyer.

LeanSheng, 2024-12-19:
@nate-desimone i will ask Nicholas to reopen this issue or I will file a new one. Since you are no longer in the FSP/BIOS team, by right you should give back the admin rights of this repo? Or else the public will have the false impression they you are acting on official term which is not right.

miczyg1, 2024-12-19:
> You can redistribute the tools and instructions described in that article, but not the resulting image.

@nate-desimone does it mean that one is not be able to redistribute EDK2 MinPLatform resulting image nor Slimbootlaoder image?

MinPlatfrom uses a rebase:
https://github.com/tianocore/edk2-platforms/blob/master/Platform/Intel/build_bios.py#L386
Slimbootloader uses split and rebase: 
https://github.com/slimbootloader/slimbootloader/blob/master/BuildLoader.py#L1002
https://github.com/slimbootloader/slimbootloader/blob/master/BuildLoader.py#L1003