intel/FSP #114: Release date for BirchStream

memristor2, 2025-01-21:
Any hints on when the FSP package for BirchStream would be released on this repo?

memristor2, 2025-01-21:
@pp3345 
Yes I have already seen that and tried using it but I don't know why the execution doesn't go any further than 0x4 post code in Fsp-T. I have checked the TemporaryRamBase and Size plus the FspTUpdParams and used the proper microcodes. I don't know what else is needed? maybe the FspListLibNull.inf file which I don't have? although I'm not sure whether the FspSecCore-T uses these pcds or not. Anyways the initTempRamApi routine never gets to the 0x7F post code (an indicator that the InitTempRamApi is done). That is why I asked for the official relase date for this FSP package since i thought I might be missing other resources

walterchris, 2025-01-21:
Do you need to use FSP-T? What host firmware are you running? coreboot?

walterchris, 2025-01-22:
coreboot has an open-source CAR implementation thus on some (most) platforms FSP-T is optional. That's why I am asking. Without source code access it's hard to debug. Without an NDA I can't tell you more on this.

Feel free to reach out to me if you need professional help on this.

memristor2, 2025-01-22:
@walterchris Thanks for the helpful information.

closed 2025-01-22 by memristor2

memristor2, 2025-01-29:
@walterchris I tried contacting you with the email you have on your git profile. I would definitely appreciate the help