intel/FSP #116: MTL IoT FSP does not set up DPR

miczyg1, 2025-03-10:
The MTL IoT FSP 4122_56 does not program the DPR register. When attempting to intialile Intel TXT with the aforementioned FSP, the DPR check in coreboot yields an error that DPR is not programmed:

```
```

miczyg1, 2025-03-10:
Actually, it can't even be workarounded. When attempted to program the unlocked DPR before FSP MemoryInit, the top of DPR (bits 31:20 of the register) remain unchanged, while EPM and DPR size are getting updated. I presume the PDR programming must happen before certain MRC steps?

pietrushnic, 2025-07-22:
@nate-desimone, any chance someone from Intel can look at it?

nate-desimone, 2025-07-22:
@miczyg1, @pietrushnic - It looks like DPR programming got moved to the NotifyPhase(EnumInitPhaseAfterPciEnumeration) API. Is your bootloader calling that API?